itead/ITEADLIB_Arduino_WeeESP8266 #3: Pinout

wifixcort, 2015-02-21:
It would be good to include the pinout in the readme

Phyks, 2015-02-24:
:+1: It was part of the [old lib](https://github.com/itead/ITEADLIB_Arduino_ESP8266) and it was way much easier to start with these chips using the old lib :)

wifixcort, 2015-02-25:
I thought the same, but if someone does not know the old library and starts
with the new library, could immediately discard it for not knowing the
pinout.

It's just a suggestion :)

prc7965, 2015-02-25:
But i like the way with old lib. It supports more boards & u can just leave
the RTX jumpers when uploading code.
Hmm, maybe i'm misunderstanding that?
Sorry 4 my bloody poor English, i learned from my gym teacher. LOL

shennongmin, 2015-02-26:
Thanks for attentions from wifixcort, Phyks and prc7965!
We have add pinout in readme for default examples. If you need to specify another serial for communication with ESP8266, modifying the code and the hardware connection are so easy.

closed 2015-02-26 by shennongmin