itead/ITEADLIB_Arduino_WeeESP8266 #19: waitForStable()

zenmanenergy, 2015-04-01:
This could be useful method by itself. You had it in the restart method. Similar to the kick() method, except it just keeps waiting. Every once in a while my board just seems to lock up, it would be nice if we could make it auto-reset itself. I haven't gotten that far yet. :-)

bool ESP8266::waitForStable(void)
{
	unsigned long start = millis();
    while (millis() - start < 3000) {
        if (eAT()) {
            delay(1500); /* Waiting for stable */
            return true;
        }
        delay(100);
    }
}

zenmanenergy, 2015-04-01:
Hmmm. This one isn't great.

I kind of think we need a way to do a hard reset. This board seems to lock all the friggin time. Maybe if we add a GPIO for the CH_PD pin to the constructor. Then if there is a need to do a hard reset we can just change the ch_pd to LOW, add a delay, then set it to HIGH. It'll need a 3.3v level shifter for that to work. I'm not even certain if cycling ch_pd really does a hard reset.

cszp17, 2015-04-10:
Thank you for your advice, we'll consider it!

closed 2015-04-10 by cszp17