itead/ITEADLIB_Arduino_WeeESP8266 #22: ESP8266 wifi(mySerial);

wifixcort, 2015-04-01:
I'm  using an arduino nano v3 and I got this error

ConnectWiFi:28: error: no matching function for call to ‘ESP8266::ESP8266(SoftwareSerial&)’
/home/ricardo/.AIDE/arduino/libraries/WeeESP8266-master/ESP8266.h:37: note: candidates are: ESP8266::ESP8266(HardwareSerial&)
/home/ricardo/.AIDE/arduino/libraries/WeeESP8266-master/ESP8266.h:29: note:                 ESP8266::ESP8266(const ESP8266&)

If I just write

ESP8266 wifi(Serial);

Apparently my problem has gone but I can't connect the ESP with my AP

Any idea?

zenmanenergy, 2015-04-03:
Does this happen every time you try? I've found with my esp01 it locks up from time to time. I found that I if I cycle power to the esp it can connect again. I'm experiment with using a transistor inline with the 3v3 power on the esp board and then allowing for cycling the power from my arduino. If/when I figure it out I'll try and add a method into the library to handle this process automatically.

wifixcort, 2015-04-03:
No this happen always, is not ESP problem, is a library problem. why I think this?. 
In the readme says if a have an UNO (nano is basically the same), add this

#include <SoftwareSerial.h>
SoftwareSerial mySerial(3, 2); /* RX:D3, TX:D2 */
ESP8266 wifi(mySerial);

If I change this in the original files I have the problem just compiling 

The original files come with

ESP8266 wifi(Serial1);//This not work neither

But if I change just this line a write all compile good without need SoftwareSerial library
ESP8266 wifi(Serial);

aniston, 2015-04-06:
Hi, you need to modify the ESP8266.h file first.
-----------------------------------------------------------
Using SoftwareSerial
If you want to use SoftwareSerial to communicate with ESP8266, you need to modify the line in file ESP8266.h:

//#define ESP8266_USE_SOFTWARE_SERIAL
After modification, it should be:
#define ESP8266_USE_SOFTWARE_SERIAL
------------------------------------------------------------
include the <SoftwareSerial.h> file like this:
#include "ESP8266.h"
#include <SoftwareSerial.h>

SoftwareSerial Serial1(3, 2); /* RX:D3, TX:D2 */
ESP8266 wifi(Serial1);

PS: I use level shifters from 5V to 3.3V as my Arduino ProMINI clone is running at 5V, remember that ESP8266 is ONLY 3.3V tolerant !

regards,
..aniston

cszp17, 2015-04-10:
aniston's answer is correct! Thank you !

closed 2015-04-10 by cszp17

ushyeon28, 2019-05-18:
aniston. you are the God.