itead/ITEADLIB_Arduino_WeeESP8266 #23: error: no matching function for call to 'ESP8266::ESP8266(SoftwareSerial&)'

DeanZilla, 2015-04-12:
Hi,

I'm a noob, so forgive me if this isn't the right place to post.

I have an UNO and are using SoftwareSerial to connect to my ESP8266.  

When I compile, I get the following error "error: no matching function for call to 'ESP8266::ESP8266(SoftwareSerial&)'" when it reaches the line below "ESP8266 wifi(mySerial);"

Below is a snippet of my code.

-----Snip-----
#include "ESP8266.h"

#define SSID        "farm"
#define PASSWORD    "sheep"

#define USER_SEL_VERSION  VERSION_18
#define ESP8266_USE_SOFTWARE_SERIAL

#include <SoftwareSerial.h>
SoftwareSerial mySerial(2, 3); // RX, TX

ESP8266 wifi(mySerial);

void setup(void)
----Snip-----

Regards

Dean

blfuentes, 2015-04-12:
Check again the documentation:
https://github.com/itead/ITEADLIB_Arduino_WeeESP8266#using-softwareserial
Did you uncomment the lines it says?

https://github.com/itead/ITEADLIB_Arduino_WeeESP8266#uno
In your code(at least what you posted) you have the following:
#include
SoftwareSerial mySerial(2, 3); // RX, TX

Notice that you are missing the "included library" in the statement  #include
It should be 

#include <SoftwareSerial.h>
SoftwareSerial mySerial(3, 2); /* RX:D3, TX:D2 */

TinajaLabs, 2015-04-14:
I was having the same problem.  

The readme says I should uncomment this in ESP8266.h 
```
//#define ESP8266_USE_SOFTWARE_SERIAL
```
but the command is not even in ESP8266.h.

Instead I had to add it.

It says to change this line:
```
#define SERIAL_BUFFER_SIZE 64
```
but instead I only see references to:
```
#define SERIAL_TX_BUFFER_SIZE 64
#define SERIAL_RX_BUFFER_SIZE 64
```

The top of my file defined for arduino uno is:

```
#include "ESP8266.h"
#include <SoftwareSerial.h>

SoftwareSerial mySerial(3, 2); /* RX:D3, TX:D2 */
ESP8266 wifi(mySerial);

#define SSID        "myssid"
#define PASSWORD    "mypasswd"
#define HOST_NAME   "node-red"
#define HOST_PORT   (4894)
```

No joy...  It still does not work.

I would like to make it work with Arduino Micro.

cszp17, 2015-04-14:
Has been updated. Thank you for your questions！

cszp17, 2015-04-17:
@TinajaLabs  We have used the 0.18 version of the firmware, the latest version of the library, switch to the 0.18 version of the UNO test software, serial port, has been running. You can test it, thank you for your question!

closed 2015-04-17 by cszp17

milrak, 2016-05-16:
Aqui está a solução para o problema @TinajaLabs 

https://github.com/itead/ITEADLIB_Arduino_WeeESP8266#using-softwareserial

Esse passo corrige o erro -> error: no matching function for call to 'ESP8266::ESP8266(SoftwareSerial&)'