itead/ITEADLIB_Arduino_WeeESP8266 #27: Create TCP Error?

weihan1102, 2015-04-25:
Hi, thanks for the great library, but i have a problem using it.

I test my board with the example "HTTPGET" . But i got an error. "create tcp err". 

I have changed www.baidu.com to a local server running on my computer but still get the same error.

Here is the log:

setup begin
FW Version:AT+GMR
00200.9.5(b1)
compiled @ Dec 25 2014 21:40:28
AI-THINKER Dec 25 2014

OK

to station + softap ok
Join AP success
IP:AT+CIFSR
+CIFSR:APIP,"192.168.4.1"
+CIFSR:APMAC,"1a:fe:34:a2:ff:e3"
+CIFSR:STAIP,"192.168.1.5"
+CIFSR:STAMAC,"18:fe:34:a2:ff:e3"

OK

single ok
setup end
create tcp err
release tcp err

closed 2015-04-25 by weihan1102

ronan-fontenay, 2015-04-30:
Hello Weihan1102, 

I have the same problem, can you inform me about what you did ?

Thank you !

jmas, 2015-05-23:
Hello, did you solve problem? I have the same.

umairyounus, 2015-07-29:
I'm getting the same problem

Marcel-G, 2015-08-18:
Also having the same problem. Any ideas on this one?

umairyounus, 2015-08-18:
Marcel-G,

I solved my problem by deleted all esp8266 libraries and by enabling software serial.

Cheers

Umair