itead/ITEADLIB_Arduino_WeeESP8266 #28: Failure while using the example

ngalongc, 2015-04-26:
Thank you for your library, you are very kind for the arduino community.
Here is the issue I faced while using the library, I was fine when I use standalone AT command.
Like the one below, it is fully functional.
```
AT version:0.21.0.0
SDK version:0.9.5

```
Yet, when I am using the library, it all goes err.
Mine is Arduino Uno, ESP8266 version as above.
Is it my hardward problem or should I update my firmware?
```
AT+GMR
AT+CWMODE?
to station err
AT+CWJAP="SSID","PASSWORD"
Join AP failure
setup end
```

cszp17, 2015-05-04:
@ngalongc 
Hi ngalongc,
    The UNO connected to ESP8266 by software serial.So you have to open a macro: 
      //#define ESP8266_USE_SOFTWARE_SERIAL  ,And that was explained in README,

cszp17, 2015-05-04:
@ngalongc 
Hi ngalongc,
The UNO connected to ESP8266 by software serial.So you have to open a macro in ESP8266 .h: 
//#define ESP8266_USE_SOFTWARE_SERIAL ,And that had be explained in README,
you can go to have a look !

closed 2015-05-04 by cszp17