itead/ITEADLIB_Arduino_WeeESP8266 #31: Why didn't you implement the Arduino WiFi Shield API?

KMax, 2015-06-24:
This is a question and it's also related to #30 issue. 

There is an MQTT implementation for ESP8266 and _(probably it also can be used for)_ Arduino, it's based on [the WiFi Shield API](https://github.com/Imroy/pubsubclient), as I understand. Also as I see people try to implement the WiFi Shield API to make their libraries compatible with other existing libraries which use WiFi.

The question is why did you decide to create your [own API](https://github.com/itead/ITEADLIB_Arduino_WeeESP8266#api-list) instead of implementing the WiFi Shield API which is a de-facto standard for Arduino, as I see it?

In theory, _I can be wrong_, if this library would has the WiFi Shield API to could be easy to use the mentioned MQTT implementation.

ricardojlrufino, 2015-10-05:
You can follow this interface
https://github.com/esp8266/Arduino/blob/esp8266/hardware/esp8266com/esp8266/doc/reference.md#wifiesp8266wifi-library

https://github.com/esp8266/Arduino/tree/esp8266/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src

a-andreyev, 2015-10-05:
@ricardojlrufino, this question is outdated a bit
Take a look at: https://github.com/semiotproject/SparkFun_ESP8266_AT_Arduino_Library (with @KMax we've tried to fork SparkFun library to add udp support, so udp support is very alpha, not tested much)

geovanisouza92, 2015-11-26:
+1 Please, create a new or adapt this lib as Client/Server interfaces for Arduino. This would be a game changing for Arduino to just add the ESP8266 instead of an expensive Wifi/EthernetShield, and will enable all Arduino's to communicate with Bluemix, RabbitMQ, Node-RED, etc.

geovanisouza92, 2015-11-26:
Consider using Stream instead of [Hardware|Software]Serial.

iddq, 2016-02-13:
@ricardojlrufino
Can you please update the urls in your comment?
https://github.com/itead/ITEADLIB_Arduino_WeeESP8266/issues/31#issuecomment-145567825

ricardojlrufino, 2016-02-13:
https://github.com/esp8266/Arduino/blob/master/doc/libraries.md#wifiesp8266wifi-library

https://github.com/esp8266/Arduino/tree/master/libraries/ESP8266WiFi

iddq, 2016-02-14:
@ricardojlrufino: thanks, is it following the WiFi Shield API?