itead/ITEADLIB_Arduino_WeeESP8266 #35: UDP

Smart-Dragon, 2015-07-26:
Help me, please!
Code in standart WiFi Arduino module for UDP:

void setup() {
   ...
   Udp.begin(localPort);
}
void loop() {

  // if there's data available, read a packet
  int packetSize = Udp.parsePacket();
  if (packetSize){ ... }

}

How to use your library to listen to packets to a specific port?

a-andreyev, 2015-07-26:
first of all you should specify what's the version of the firmware flashed on you esp8266 module

Smart-Dragon, 2015-07-26:
FW Version:AT version:0.25.0.0(Jun  5 2015 16:27:16)
SDK version:1.1.1
Ai-Thinker Technology Co. Ltd.
Jun  5 2015 23:07:20

a-andreyev, 2015-07-26:
I would recommend to try my fork #32 for the firmwares version higher than v0.22 for apropriate udp support.
With this implementation you could try something like in the .../examples/.../UDPClientSingle.ino to get UDP server working.
Good luck :)

Smart-Dragon, 2015-07-26:
Everything works! Thanks!

closed 2015-08-03 by cszp17