itead/ITEADLIB_Arduino_WeeESP8266 #36: Serial1' was not declared in this scope

teinhonglo, 2015-07-27:
Sorry I have two question

(1).I run the sample code  "Connect Wifi", but complier error "Serial1' was not declared in this scope".

(2).The circuit isn't it?
http://microcontrollerelectronics.com/wp-content/uploads/2014/12/esp8266_bb.png

a-andreyev, 2015-07-27:
Scheme is ok in general, but I'm not sure you could easy use Leonardo Serial1 for communication:
> The Arduino Leonardo board uses Serial1 to communicate via TTL (5V) serial on pins 0 (RX) and 1 (TX). Serial is reserved for USB CDC communication.

As an option, try to test it with SoftwareSerial

teinhonglo, 2015-07-27:
sorry i link png is wrong, i use arduino, as below
http://rancidbacon.com/images/project/esp8266-doc/ESP8266-Connections_bb-fixed-small.png

and i just compiler "Connect Wifi" , then output is
http://i.imgur.com/4DSaKRe.jpg

a-andreyev, 2015-07-28:
Arduino Uno doesn't have Serial1, only Serial:
> Serial: 0 (RX) and 1 (TX). Used to receive (RX) and transmit (TX) TTL serial data. These pins are connected to the corresponding pins of the ATmega8U2 USB-to-TTL Serial chip.

If you are going to use Software Serial as described:
```
#include "ESP8266.h"
#include <SoftwareSerial.h>

SoftwareSerial mySerial(3, 2); // uno sw RX(pin3) --> esp8266 TX, uno sw TX(pin2) --> esp8266 RX
ESP8266 wifi(mySerial);
```

You should connect pins 3 and 2 (software rx and tx) to esp8266 tx and rx pins (in reverse mode: rx to tx) according to pinout:
[image] 
So you should move green and yellow wires in your uno from serial pins (0 and 1) to 2nd and 3rd pin. Your second scheme doens't fit, the first scheme with leonardo was more clear about rx-tx connection for sending commands to esp8266 in regular mode.

Also don't forget to initialize your software serial in setup function:
```
void setup() {
mySerial.begin(9600); // or some other apropriate baudrate depending on firmware
...
}
```

Good luck!

teinhonglo, 2015-07-28:
Sorry, I also cannot run it....
http://i.imgur.com/zkZdEyk.jpg

my pins 3 and 2 is software rx and tx

a-andreyev, 2015-07-28:
SoftwareSerial mySerial(3, 2), so:
> pin3 (software RX) --> esp8266 TX
> pin2 (software TX) --> esp8266 RX,

right?

As I can see, failure present as eary stage when sending esp8266 working mode.

Try to add something like this to check what stages are not passing:
```
void setupESP8266()
{
    while (1) {
        Serial.print("restaring esp8266...");
        if (esp8266.restart()) {
            Serial.print("ok\r\n");
            break;
        }
        else {
            Serial.print("not ok...\r\n");
            Serial.print("Trying to kick...");
            while (1) {
                if (esp8266.kick()) {
                    Serial.print("ok\r\n");
                    break;
                }
                else {
                    Serial.print("not ok... Wait 5 sec and retry...\r\n");
                    delay(5000);
                }
            }
        }
    }

    Serial.print("setup begin\r\n");
    Serial.print("FW Version:");
    Serial.println(esp8266.getVersion().c_str());

    while (1) {
        if (esp8266.setOprToStation()) {
            Serial.print("to station ok\r\n");
            break;
        }
        else {
            Serial.print("to station err\r\n");
            Serial.println("Wait 5 seconds and try again...");
            delay(5000);
        }
    }

    while (1) {
        if (esp8266.joinAP(SSID, PASSWORD)) {
            Serial.print("Join AP success\r\n");
            break;
        }
        else {
            Serial.print("Join AP failure\r\n");
            Serial.println("Wait 5 seconds and try again...");
            delay(5000);
        }
    }
    Serial.print("setup end\r\n");
}

void setup() {
 ... // serial setup, etc
 setupESP8266();
}
```

teinhonglo, 2015-07-28:
but complier error esp8266 was not declare this scope?
variable esp8266 is?

my code as below..
------------------------------------------------------------------------------------
#include "ESP8266.h"
#include <SoftwareSerial.h>

#define my_SSID        "K2F01"
#define PASSWORD       "12345678"

#include <SoftwareSerial.h>
SoftwareSerial mySerial(3, 2); /* RX:D3, TX:D2 */
ESP8266 wifi(mySerial);

void setupESP8266()
{
    while (1) {
        Serial.print("restaring esp8266...");
        if (esp8266.restart()) {
            Serial.print("ok\r\n");
            break;
        }
        else {
            Serial.print("not ok...\r\n");
            Serial.print("Trying to kick...");
            while (1) {
                if (esp8266.kick()) {
                    Serial.print("ok\r\n");
                    break;
                }
                else {
                    Serial.print("not ok... Wait 5 sec and retry...\r\n");
                    delay(5000);
                }
            }
        }
    }

    Serial.print("setup begin\r\n");
    Serial.print("FW Version:");
    Serial.println(esp8266.getVersion().c_str());

    while (1) {
        if (esp8266.setOprToStation()) {
            Serial.print("to station ok\r\n");
            break;
        }
        else {
            Serial.print("to station err\r\n");
            Serial.println("Wait 5 seconds and try again...");
            delay(5000);
        }
    }

    while (1) {
        if (esp8266.joinAP(SSID, PASSWORD)) {
            Serial.print("Join AP success\r\n");
            break;
        }
        else {
            Serial.print("Join AP failure\r\n");
            Serial.println("Wait 5 seconds and try again...");
            delay(5000);
        }
    }
    Serial.print("setup end\r\n");
}

void setup() {
    // serial setup, etc
    setupESP8266();
    mySerial.begin(9600);
    Serial.begin(9600);
    Serial.print("setup begin\r\n");
    Serial.println(wifi.kick());
    Serial.print("FW Version: ");
    Serial.println(wifi.getVersion().c_str());
    
    if (wifi.setOprToStation()) {
        Serial.print("to station ok\r\n");
    } else {
        Serial.print("to station err\r\n");
    }

    if (wifi.joinAP(my_SSID, PASSWORD)) {
        Serial.print("Join AP success\r\n");
        Serial.print("IP: ");       
        Serial.println(wifi.getLocalIP().c_str());
    } else {
        Serial.print("Join AP failure\r\n");
    }
    
    Serial.print("setup end\r\n");
}

void loop(void)
{
}

teinhonglo, 2015-07-28:
Sorry, My RX and TX is reserve, I exchange them, that is OK!(so shame...Orz)
but it output unexpected message, that is correct?
http://i.imgur.com/pjhzFYw.jpg

a-andreyev, 2015-07-28:
Ok, not problem! I'm glad that it more or less working at least :)

About the serial monitor results:
looks wrong, like some interference noise in the wires or something.
Not sure, it's hard to predict but looks not good for me.

umairyounus, 2015-07-29:
Hi,

I'm getting this error:

ConnectWiFi.cpp.o: In function `__static_initialization_and_destruction_0':
/Users/umair/ConnectWiFi.ino:7: undefined reference to `ESP8266::ESP8266(SoftwareSerial&, unsigned long)'
collect2: error: ld returned 1 exit status
Error compiling.

Here is the code:

#define ESP8266_USE_SOFTWARE_SERIAL
#include "ESP8266.h"
#include <SoftwareSerial.h>

SoftwareSerial mySerial(3, 2); // RX, TX

ESP8266 wifi(mySerial);

#define SSID        "ITEAD"
#define PASSWORD    "12345678"

void setup(void)
{
    mySerial.begin(9600); // or some other apropriate baudrate depending on firmware
    Serial.begin(9600);
    Serial.print("setup begin\r\n");

    Serial.print("FW Version: ");
    Serial.println(wifi.getVersion().c_str());
    
    
    if (wifi.setOprToStation()) {
        Serial.print("to station ok\r\n");
    } else {
        Serial.print("to station err\r\n");
    }

    if (wifi.joinAP(SSID, PASSWORD)) {
        Serial.print("Join AP success\r\n");
        Serial.print("IP: ");       
        Serial.println(wifi.getLocalIP().c_str());
    } else {
        Serial.print("Join AP failure\r\n");
    }
    
    Serial.print("setup end\r\n");
}

void loop(void)
{
}

cszp17, 2015-08-03:
You are using a soft serial port,you should to confirm that:
1:the wiring is correct
2:open the macro in ESP8266.h ?  ->  #define ESP8266_USE_SOFTWARE_SERIAL

closed 2015-08-03 by cszp17

inflamers, 2015-09-25:
I can't get it working, wiring is good, I switched RX & TX, on PIN 3 & 2. No error while compiling.
But it doesn't connect 
setup begin
FW Version: 
to station err
Join AP failure
setup end

Any idea ?

DriesX, 2015-09-25:
What baudrate are you using?
I had to test a few before it worked. According to docs I found it had to be 9800 but it also didn't work.

DriesX, 2015-09-25:
I'm now using 115200 and it works.

inflamers, 2015-09-25:
I was using 9600, I tried with each ones (including 115200) nothing works, I keep having these errors...

Here is my code 

#include <SoftwareSerial.h>
#include <doxygen.h>
#include <ESP8266.h>

SoftwareSerial mySerial(3, 2); /* RX:D3, TX:D2 */
ESP8266 wifi(mySerial);

#define SSID        "ssid"
#define PASSWORD    "passwd"

void setup(void)
{

    Serial.begin(115200);
    Serial.print("setup begin\r\n");

    Serial.print("FW Version: ");
    Serial.println(wifi.getVersion().c_str());
    
    if (wifi.setOprToStation()) {
        Serial.print("to station ok\r\n");
    } else {
        Serial.print("to station err\r\n");
    }

    if (wifi.joinAP(SSID, PASSWORD)) {
        Serial.print("Join AP success\r\n");
        Serial.print("IP: ");       
        Serial.println(wifi.getLocalIP().c_str());
    } else {
        Serial.print("Join AP failure\r\n");
    }
    
    Serial.print("setup end\r\n");
}

void loop(void)
{
}

inflamers, 2015-09-25:
I also don't use any level shifter...

DriesX, 2015-09-25:
You have to use the two serials.

Serial: console: all commands to print should be Serial.print(...)
mySerial for esp8266.

Both Serials should have the begin statement.

Serial.begin(9600); //for your print
mySerial.begin(115200); //for wifi (baudrate depends)

inflamers, 2015-09-25:
Correct, I forgot to put it back, but still no response...

#include <SoftwareSerial.h>
#include <doxygen.h>
#include <ESP8266.h>

SoftwareSerial mySerial(3, 2); /* RX:D3, TX:D2 */
ESP8266 wifi(mySerial);

#define SSID        "ssid"
#define PASSWORD    "paswd"

void setup(void)
{
    mySerial.begin(115200);
    Serial.begin(9600);
    Serial.print("setup begin\r\n");
    Serial.print("FW Version: ");
    Serial.println(wifi.getVersion().c_str());
    
    if (wifi.setOprToStation()) {
        Serial.print("to station ok\r\n");
    } else {
        Serial.print("to station err\r\n");
    }

    if (wifi.joinAP(SSID, PASSWORD)) {
        Serial.print("Join AP success\r\n");
        Serial.print("IP: ");       
        Serial.println(wifi.getLocalIP().c_str());
        
    } else {
        Serial.print("Join AP failure\r\n");
    }
    
    Serial.print("setup end\r\n");
}

inflamers, 2015-09-25:
Correct I forgot this line,

It seems to work better than last time 

FW Version: 
to station err
Join AP success
IP: 
setup end

For some reason, I still get an "err" message, and no IP is printed, but the "Join AP success" is pretty encouraging ! Many thanks for your help !

next step is sending an email !

inflamers, 2015-09-25:
and while running a lan scan, the device appears, so it seems to be really connected, great !

Vigneshramkumar, 2016-02-14:
Hi,
I am trying to connect ESP8266 and Arduino UNO. I downloaded the ESP8266 library and tried the ConnectWifi program in it. But I am not able to connect to the wifi network. It is giving me the following error.
setup begin
FW Version: 
to station err
Join AP failure
setup end

PFB the program I have uploaded:
#include "ESP8266.h"
#include "SoftwareSerial.h"

SoftwareSerial mySerial(3,2);
ESP8266 wifi(mySerial);

#define SSID        "--------"
#define PASSWORD    "------------------"
//#define SERIAL_BUFFER_SIZE 256
//#define SERIAL_RX_BUFFER_SIZE 64

//ESP8266 wifi(Serial);
void setup(void)
{
    Serial.begin(9600);
    mySerial.begin(115200); 
    Serial.print("setup begin\r\n");

    Serial.print("FW Version: ");
    Serial.println(wifi.getVersion().c_str());
    
    
    if (wifi.setOprToStation()) {
        Serial.print("to station ok\r\n");
    } else {
        Serial.print("to station err\r\n");
    }

    if (wifi.joinAP(SSID, PASSWORD)) {
        Serial.print("Join AP success\r\n");
        Serial.print("IP: ");       
        Serial.println(wifi.getLocalIP().c_str());
    } else {
        Serial.print("Join AP failure\r\n");
    }
    
    Serial.print("setup end\r\n");
}

void loop(void)
{
}
Kindly someone help me to connect the shield with the network. I already made some research with other issues section in Git hub but nothing happened.
Thanks in advance.

agrawalnikhil, 2016-02-18:
I am using the above code 
`#include <doxygen.h>
#include <ESP8266.h>
#include <SoftwareSerial.h>
#define SSID        "NOKIA Lumia 520_2453"
#define PASSWORD    "70782447"
SoftwareSerial mySerial(3, 2); /* RX:D3, TX:D2 */
ESP8266 wifi(mySerial);
void setupesp8266()
{
     while (1) {
        Serial.print("restaring esp8266...");
        if (esp8266.restart()) {
            Serial.print("ok\r\n");
            break;
        }
        else {
            Serial.print("not ok...\r\n");
            Serial.print("Trying to kick...");
            while (1) {
                if (esp8266.kick()) {
                    Serial.print("ok\r\n");
                    break;
                }
                else {
                    Serial.print("not ok... Wait 5 sec and retry...\r\n");
                    delay(5000);
                }
            }
        }
    }
    Serial.begin(9600);
   mySerial.begin(115200);
    Serial.print("setup begin\r\n");
  Serial.print("FW Version: ");
    Serial.println(esp8266.getVersion().c_str());
 if (esp8266.setOprToStation()) {
        Serial.print("to station ok\r\n");
        break;
    } else {
        Serial.print("to station err\r\n");
          Serial.println("Wait 5 seconds and try again...");
            delay(5000);
    }
    while(1){
    if (esp8266.joinAP(SSID, PASSWORD)) {
        Serial.print("Join AP success\r\n");
        Serial.print("IP: ");       
        Serial.println(wifi.getLocalIP().c_str());
        break;
    } else {
        Serial.print("Join AP failure\r\n");
        Serial.println("Wait 5 seconds and try again...");
            delay(5000);
    }
   
    }
           Serial.print("setup end\r\n");

}
void setup(void)
{
  setupesp8266();
}
void loop(void)
{
}
`
but is giving following error

ConnectWiFi:12: error: 'esp8266' was not declared in this scope

         if (esp8266.restart()) {

             ^

ConnectWiFi:35: error: 'esp8266' was not declared in this scope

     Serial.println(esp8266.getVersion().c_str());

                    ^

ConnectWiFi:38: error: break statement not within loop or switch

         break;

         ^

Using library ESP8266Wifi in folder: C:\Users\admin\Documents\Arduino\libraries\ESP8266Wifi (legacy)
Using library SoftwareSerial at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SoftwareSerial 
exit status 1
'esp8266' was not declared in this scope

please help me

alanbennett, 2017-02-22:
Can anyone help?

Arduino UNO 
genuino 1.8.1

※※※※※※※※※※※※※※※※※※※※
UNO  ->  ESP8266
----------------------------------------
3.3v    ->  3V3
3.3v   ->  EN
GND  -> GND
D2      -> TX
D3     ->  RX
※※※※※※※※※※※※※※※※※※※※

**## MY CODE ####################**
#include "ESP8266.h"
#include "SoftwareSerial.h"

#define SSID        "cgun"
#define PASSWORD    "123456789"

SoftwareSerial mySerial(3, 2); // RX,TX     uno sw RX(pin3) --> esp8266 TX, uno sw TX(pin2) --> esp8266 RX
ESP8266 wifi(mySerial);

void setup(void)
{
    Serial.begin(9600);
  mySerial.begin(115200); 

    Serial.print("setup begin\r\n");
    Serial.print("FW Version: ");
    Serial.println(wifi.getVersion().c_str());

    if (wifi.setOprToStation()) {
        Serial.print("to station ok\r\n");
    } else {
        Serial.print("to station err\r\n");
    }

    if (wifi.joinAP(SSID, PASSWORD)) {
        Serial.print("Join AP success\r\n");
        Serial.print("IP: ");       
        Serial.println(wifi.getLocalIP().c_str());
    } else {
        Serial.print("Join AP failure\r\n");
    }
    Serial.print("setup end\r\n");
}

void loop(void)
{
}
**################################**

**## RESULT ####################**
setup begin
FW Version: 
to station err
Join AP failure
setup end
**################################**

[image]

JMAN2018, 2018-02-26:
I having a problem uploading my sketch to my attinty 85. it says serial was not declared this scope. heres my code. what do I need to d to get it to upload. I'm also not sure about the in and out pins for this. its the ldd sensor as out and light bulb as the input.

int LDR = 0;     //analog pin to which LDR is connected, here we set it to 0 so it means A0
int LDRValue = 0;      //that’s a variable to store LDR values
int light_sensitivity = 500;    //This is the approx value of light surrounding your LDR
 
void setup()
  {
    Serial.begin(9600);          //start the serial monitor with 9600 buad
    pinMode(7, OUTPUT);     //we mostly use 13 because there is already a built in yellow LED in arduino which shows output when 13 pin is enabled
  }
 
void loop()
  {
    LDRValue = analogRead(LDR);      //reads the ldr’s value through LDR 
-    Serial.println(LDRValue);       //prints the LDR values to serial monitor

}
    delay(50);        //This is the speed by which LDR sends value to arduino
 
    if (LDRValue < light_sensitivity) 
      {
        digitalWrite(7, HIGH);
      }
    else
      {
        digitalWrite(7, LOW);
      }
  }

itxopps, 2018-05-11:
hi there 
i am new to programing. i have prepared a code for my robot. i worrks completly i compiled it many time uploaded without any problem eddited many times but suddenly it genrated an error. SERIALL1 is not decleared in this scop.i am using hc12 rf module with arduino mega 2560. code error is here.......
[image]
the complete code is here.

#include <LiquidCrystal.h>

// initialize the library with the numbers of the interface pins
LiquidCrystal lcd(12, 11, 9, 8, 7, 6);
#include <Servo.h>

#include <HCPCA9685.h>
#define  I2CAdd 0x40
HCPCA9685 HCPCA9685(I2CAdd);

#define trigPin1 A5
#define echoPin1 A6

char val;// variable to receive data from the serial port

int light= 53;

int motor1Pin1 = 2; // pin 2 on L293D IC
int motor1Pin2 = 3; // pin 7 on L293D IC
int motor2Pin1 = 4; // pin 10 on L293D IC
int motor2Pin2 = 5; // pin 15 on L293D IC

int in1 = A1;
int in2 = A2;
int in3 = A3;
int in4 = A4;

int cont1 = 0;
int cont2 = 0;
int cont3 = 0;
int cont4 = 0;

// Array to store pins nos
int remotePins [] = {A1, A2, A3, A4 };

const int sensor=A0;
float tempc; 
float vout; 

int servo1 = 10; //10 to 420 
int servo2 = 200; //50 to 420
int servo3 = 420;//420
int servo4 = 150;
/
int servo5 = 220; //220  to 350

int servo6 = 420;//10 to 420

int servo7 = 200;
int servo8 = 20;//10 to 420
int servo9 = 150;

int servo10 = 250;

long duration, distance,Sensor;

void setup() 

{

  for (int i = 0; i < 5; i ++) 
  {
    pinMode (remotePins [i], INPUT);
    digitalWrite(remotePins [i], HIGH);
  }
[... 352 lines cut ...]
cont4 = 0;  
delay(100); 
 }
 }

//Serial.print("in DegreeC=");
//Serial.print("\t");
//Serial.println(tempc);
 
}

void lightON(){digitalWrite(light, HIGH);}
void lightOFF(){digitalWrite(light, LOW);}

void SonarSensor(int trigPin,int echoPin)
{
digitalWrite(trigPin, LOW);
delayMicroseconds(2);
digitalWrite(trigPin, HIGH);
delayMicroseconds(10);
digitalWrite(trigPin, LOW);
duration = pulseIn(echoPin, HIGH);
distance = (duration/2) / 29.1;

}