itead/ITEADLIB_Arduino_WeeESP8266 #40: NTP client

leruetkins, 2015-10-23:
Please tell me can this library connect to ntp server and get unix time?

niki-timofe, 2015-10-23:
Yes. NTP is realized by UDP, so you can look example in arduino IDE, how to issue ntp-server by UDP.

leruetkins, 2015-10-24:
I saw an example in the standard library wifi, but I do not understand how to use this library, can someone tell me the code?

leruetkins, 2015-10-24:
Find solution to get unix time with this library:

#include "ESP8266.h"
#include <SoftwareSerial.h>
SoftwareSerial espSerial(10, 11);

#define HOST_NAME   "82.209.243.241"
#define HOST_PORT   (123)

ESP8266 wifi(espSerial);

void setup()
{
  Serial.begin(9600);
  Serial.print("setup begin\r\n");
  Serial.print("Join AP success\r\n");
  Serial.print("IP: ");
  Serial.println(wifi.getLocalIP().c_str());

  if (wifi.disableMUX()) {
    Serial.print("single ok\r\n");
  } else {
    Serial.print("single err\r\n");
  }

  Serial.print("setup end\r\n");
}
void loop()
{
  ntpupdate();
  delay(20000);
}

void ntpupdate()
{
  uint8_t buffer[128] = {0};

  if (wifi.registerUDP(HOST_NAME, HOST_PORT)) {
    Serial.print("register udp ok\r\n");
  } else {
    Serial.print("register udp err\r\n");
  }

  static const char PROGMEM
  timeReqA[] = { 227,  0,  6, 236 },
               timeReqB[] = {  49, 78, 49,  52 };
  // Assemble and issue request packet
  uint8_t       buf[48];
  memset(buf, 0, sizeof(buf));
  memcpy_P( buf    , timeReqA, sizeof(timeReqA));
  memcpy_P(&buf[12], timeReqB, sizeof(timeReqB));

  //    char *buf = "Hello, this is client!";
  wifi.send((const uint8_t*)buf, 48);

  uint32_t len = wifi.recv(buffer, sizeof(buffer), 10000);
  if (len > 0) {
    Serial.print("UNIX TIME IS:");

    // Serial.print(buffer[42]);
    unsigned long t = (((unsigned long)buffer[40] << 24) |
                       ((unsigned long)buffer[41] << 16) |
                       ((unsigned long)buffer[42] <<  8) |
                       (unsigned long)buffer[43]) - 2208988800UL;

    Serial.println(t);
  }

  if (wifi.unregisterUDP()) {
    Serial.print("unregister udp ");
    Serial.println(" ok");
  } else {
    Serial.print("unregister udp ");
    Serial.println(" err");
  }
}

closed 2016-01-05 by cszp17