itead/ITEADLIB_Arduino_WeeESP8266 #41: TCP server not working after 1 connection

axwell, 2015-10-29:
Hello,
i found your great library but i'm having some issues with the TCP server.
With the given example after 1 successfully request i am unable to do a second one. No reponse whatsoever.
Do you have any idea why ?

Thank you in advance!

cszp17, 2015-10-30:
Can you give me some debug information?

axwell, 2015-10-30:
Sure!
Log file uploaded here: https://romcloud.eu/index.php/s/LrTKndG5b3kZXg7

First request is ok
-send back ok
-release tcp 0 ok

After that there is not data send back, browser ERR_EMPTY_RESPONSE.
-send back err
-release tcp0 err
Tested with chrome, mozilla, and Edge.

cszp17, 2015-11-02:
From your log ,I think that there is a question with your client code.you can try to find that why the status[] is null ,if right ,it will be Status:[STATUS:3+CIPSTATUS:1,"TCP","192.168.1.110",34062,1]

cszp17, 2015-11-02:
From your log ,I think that there is a question with your client code.you can try to find that why the status[] is null ,if right ,it will be like Status:[STATUS:3+CIPSTATUS:1,"TCP","192.168.1.110",34062,1]

cszp17, 2015-11-02:
i have a try ,and don't find question that you meet.so i think it is some questions with your client code.maybe you can try that when you do a one ,you disconnect server.then connect server when you do a second one.

closed 2015-11-02 by cszp17