itead/ITEADLIB_Arduino_WeeESP8266 #43: UNO does not work with 0.9.5 firmware

domonetic, 2015-11-20:
UNO does not work with 0.9.5 firmware and SoftwareSerial
What version of firmware used?

famaridon, 2016-01-04:
Hi, 

I got the same problem with this firmware version : 
```
AT+GMR                                                                          
AT version:0.51.0.0(Nov 27 2015 13:37:21)                                       
SDK version:1.5.0                                                               
compile time:Nov 27 2015 13:58:02                                               
OK 
```

with 0.9.5 firmware default beaud rate change from 9600 to 115200.

but Uno SoftwareSerial can't handel 115200 baud out of the box. 

You can use (for current )
```
AT+UART_CUR=38400,8,1,0,0
```
Or (for default)
```
AT+UART_DEF=38400,8,1,0,0
```

to downgrade your ESP8266 baud rate.

when i do it it's work fine for me.

domonetic, 2016-01-05:
Not. 0.9.5 use 9600, but this Library dont work with this firmware.