itead/ITEADLIB_Arduino_WeeESP8266 #48: Can't make it work with Arduino Uno

yuflowdev, 2015-12-30:
Hello,

i'm trying to plug a Wee serial wifi module (http://wiki.iteadstudio.com/Wee_Serial_WIFI_Module) on my Arduino Uno, through an Xbee shield (https://hackspark.fr/fr/xbee-shield.html) .

After some tough tryouts to make upload work again, i can now upload my sketch to the arduino (the sketch is the example from the library).

All i get is

setup begin
FW Version:
to station err
Join AP failure
setup end

What can I do ?

Thank you very much

atiwattri, 2016-01-02:
Please see this : https://github.com/itead/ITEADLIB_Arduino_WeeESP8266/issues/36

yuflowdev, 2016-01-04:
Hello,
thank you for your reply. Unfortunately, still nothing.

The code I'm using : 

`#include <SoftwareSerial.h>
#include <doxygen.h>
#include <ESP8266.h>
SoftwareSerial mySerial(3, 2); /* RX:D3, TX:D2 */
ESP8266 wifi(mySerial);

#define SSID //
#define PASSWORD //
void setup(void)
{

Serial.begin(9600);
mySerial.begin(115200);
Serial.print("setup begin\r\n");

Serial.print("FW Version: ");
Serial.println(wifi.getVersion().c_str());

if (wifi.setOprToStation()) {
    Serial.print("to station ok\r\n");
} else {
    Serial.print("to station err\r\n");
}
delay(1500);

if (wifi.joinAP(SSID, PASSWORD)) {
    Serial.print("Join AP success\r\n");
    Serial.print("IP: ");       
    Serial.println(wifi.getLocalIP().c_str());
} else {
    Serial.print("Join AP failure\r\n");
}

Serial.print("setup end\r\n");

}

void loop(void)
{
}`

Still getting the same result...

You may see the wiring here : http://i.stack.imgur.com/GsC52.jpg

atiwattri, 2016-01-04:
Hi yuflowdev, I'm not good English. But i will try help you. :-)

My firmware version is 0.9.2.2 (This version baudrate is 9600)

Wiring this
-------------------------------------------------------------------
Uno R3            ESP8266
3.3V     ---->     Vcc
Gnd     ---->     Gnd
3.3V     ---->     CH_PD
Pin3     ---->     TX
Pin2     ---->     RX
-------------------------------------------------------------------

Code
-------------------------------------------------------------------
#include <ESP8266.h>
#include <SoftwareSerial.h>

SoftwareSerial mySerial(3, 2); 
ESP8266 wifi(mySerial);

#define SSID "yourSSID"
#define PASSWORD "youPassword"

void setup(void)
{
  
    Serial.begin(9600);
    Serial.print("Setup begin\r\n");

    Serial.print("FW Version: ");
    Serial.println(wifi.getVersion().c_str());
    
    if (wifi.setOprToStation()) {
        Serial.print("to station ok\r\n");
    } else {
        Serial.print("to station err\r\n");
    }
  
    bool is_connect = wifi.joinAP(SSID, PASSWORD);
    if (is_connect) {
        Serial.println("Join AP success");
        Serial.print("IP: ");       
        Serial.println(wifi.getLocalIP().c_str());
    } else {
        Serial.println("Join AP failure\r\n");
    }
  Serial.print("setup end\r\n");
}

void loop(void)
{
}
-------------------------------------------------------------------

yuflowdev, 2016-01-04:
Hi, thank you for your time.

I'm not able to modify the wiring, because i'm using an xbee shield.

The code i'm using is the same you adviced .

Please see :
http://arduino.stackexchange.com/questions/19097/cant-make-wee-serial-wifi-work-with-arduino-uno

atiwattri, 2016-01-04:
I'm not sure this library is support Xbee shield.

Please try: http://communityofrobots.com/tutorial/kawal/xbee-and-arduino

If not work. Please let me know.

famaridon, 2016-01-04:
Hi,

read #43