itead/ITEADLIB_Arduino_WeeESP8266 #51: Can't get local ip.

matissev, 2016-01-29:
Hello,
I am having a problem with the getLocalIP() function as it only returns a white space.

Example :
```
00160901
reset ok
to access point ok
Join AP success
 <--- This is where I should get the ip, but it displays a white space instead.
multiple ok
start server ok
setup end

```

Here is my setup code.

```
#include "ESP8266.h"

#define SSID        "ssid"
#define PASSWORD    "pass"

ESP8266 wifi(Serial1);

void setup() {
	Serial.begin(115200);

	// GET FIRMWARE VERSION OF THE ESP8266
	Serial.println(wifi.getVersion().c_str());

	// RESET WIFI
	if (wifi.restart()) {
		Serial.print("reset ok\r\n");
	} else {
		Serial.print("reset failure\r\n");
	}

	// SET MODE
	if (wifi.setOprToStation()) {
		Serial.print("to access point ok\r\n");
	} else {
		Serial.print("to access point err\r\n");
	}

	// REACH ACCESS POINT
	if (wifi.joinAP(SSID, PASSWORD)) {
		Serial.print("Join AP success\r\n");
		Serial.println(wifi.getLocalIP().c_str());
	} else {
		Serial.print("Join AP failure\r\n");
	}

	// ENABLE MULTIPLE CONNECTION
	if (wifi.enableMUX()) {
		Serial.print("multiple ok\r\n");
	} else {
		Serial.print("multiple err\r\n");
	}

	// START SERVER
	if (wifi.startTCPServer(8000)) {
		Serial.print("start server ok\r\n");
	} else {
		Serial.print("start server err\r\n");
	}

	Serial.print("setup end\r\n");
}

```

Thank you for the lib by the way, despite this little thing, it works great.

iddq, 2016-02-13:
use newer firmware

esp_iot_sdk_v1.5.0_15_11_27.zip
FLASH_DOWNLOAD_TOOLS_v2.4_150924.rar

matissev, 2016-06-28:
Thank you, it worked.

closed 2016-06-28 by matissev