itead/ITEADLIB_Arduino_WeeESP8266 #52: Connecting ESP8266 + Arduino UNO

Vigneshramkumar, 2016-02-14:
Hi,
I am trying to connect ESP8266 and Arduino UNO. I downloaded the ESP8266 library and tried the ConnectWifi program in it. But I am not able to connect to the wifi network. It is giving me the following error.
setup begin
FW Version: 
to station err
Join AP failure
setup end

PFB the program I have uploaded:
#include "ESP8266.h"
#include "SoftwareSerial.h"

SoftwareSerial mySerial(3,2);
ESP8266 wifi(mySerial);

#define SSID        "--------"
#define PASSWORD    "------------------"
//#define SERIAL_BUFFER_SIZE 256
//#define SERIAL_RX_BUFFER_SIZE 64

//ESP8266 wifi(Serial);
void setup(void)
{
    Serial.begin(9600);
    mySerial.begin(115200); 
    Serial.print("setup begin\r\n");

    Serial.print("FW Version: ");
    Serial.println(wifi.getVersion().c_str());
    
    
    if (wifi.setOprToStation()) {
        Serial.print("to station ok\r\n");
    } else {
        Serial.print("to station err\r\n");
    }

    if (wifi.joinAP(SSID, PASSWORD)) {
        Serial.print("Join AP success\r\n");
        Serial.print("IP: ");       
        Serial.println(wifi.getLocalIP().c_str());
    } else {
        Serial.print("Join AP failure\r\n");
    }
    
    Serial.print("setup end\r\n");
}

void loop(void)
{
}
Kindly someone help me to connect the shield with the network. I already made some research with other issues section in Git hub but nothing happened.
Thanks in advance.

Vigneshramkumar, 2016-02-15:
Hi,

Kindly help me out for this problem. Still I am getting the same error.
I am using the following specifications:
Arduino IDE - 1.6.6
Baud rate - 9600 for serial
Wifi baud rate - 152000

Kindly response. Thanks in advace.

Regards
Vignesh

Frank0223, 2016-04-19:
@Vigneshramkumar, try the following modification:
ESP8266 wifi(mySerial, 115200);
// mySerial.begin(115200);

sjunaid9900, 2017-02-27:
Move your RX, TX pins to Pin3, Pin2