itead/ITEADLIB_Arduino_WeeESP8266 #53: esp8266 was not declared in this scope

agrawalnikhil, 2016-02-22:
I am using the above code
`#include
#include
#include
#define SSID "SSID"
#define PASSWORD "Password"
SoftwareSerial mySerial(3, 2); /* RX:D3, TX:D2 */
ESP8266 wifi(mySerial);
void setupesp8266()
{
while (1) {
Serial.print("restaring esp8266...");
if (esp8266.restart()) {
Serial.print("ok\r\n");
break;
}
else {
Serial.print("not ok...\r\n");
Serial.print("Trying to kick...");
while (1) {
if (esp8266.kick()) {
Serial.print("ok\r\n");
break;
}
else {
Serial.print("not ok... Wait 5 sec and retry...\r\n");
delay(5000);
}
}
}
}
Serial.begin(9600);
mySerial.begin(115200);
Serial.print("setup begin\r\n");
Serial.print("FW Version: ");
Serial.println(esp8266.getVersion().c_str());
if (esp8266.setOprToStation()) {
Serial.print("to station ok\r\n");
break;
} else {
Serial.print("to station err\r\n");
Serial.println("Wait 5 seconds and try again...");
delay(5000);
}
while(1){
if (esp8266.joinAP(SSID, PASSWORD)) {
Serial.print("Join AP success\r\n");
Serial.print("IP: ");

Serial.println(wifi.getLocalIP().c_str());
break;
} else {
Serial.print("Join AP failure\r\n");
Serial.println("Wait 5 seconds and try again...");
delay(5000);
}

}
       Serial.print("setup end\r\n");

}
void setup(void)
{
setupesp8266();
}
void loop(void)
{
}
`
but is giving following error

ConnectWiFi:12: error: 'esp8266' was not declared in this scope

     if (esp8266.restart()) {

         ^

ConnectWiFi:35: error: 'esp8266' was not declared in this scope

 Serial.println(esp8266.getVersion().c_str());

                ^

ConnectWiFi:38: error: break statement not within loop or switch

     break;

     ^

Using library ESP8266Wifi in folder: C:\Users\admin\Documents\Arduino\libraries\ESP8266Wifi (legacy)
Using library SoftwareSerial at version 1.0 in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\SoftwareSerial
exit status 1
'esp8266' was not declared in this scope

please help me

chinmoyrick, 2016-02-22:
Hi..
this code is error in various place.where are you get this code  

`#include ------------- include some thing (esp8266.h or wifi.h)

#define SSID "SSID"---------------------- place Here your network SSID and password
#define PASSWORD "Password"

SoftwareSerial mySerial(3, 2); /* RX:D3, TX:D2 */------------------------ Include softwareserial in esp8266.h file

void setupesp8266()--------------------- don't understand about esp8266

You have to write a clear and perfect sketch.

Nardjis, 2020-11-18:
I had this problem with the variables if their value does not fit or they are out of place, eg ledPin = D4, I simply set ledPin = 4 and that was resolved.