itead/ITEADLIB_Arduino_WeeESP8266 #54: setOprToStation() is returning False

agrawalnikhil, 2016-02-24:
#include <doxygen.h>
#include <ESP8266.h>
#include <SoftwareSerial.h>
#define SSID        "ghini"
#define PASSWORD    "counterstrike"
SoftwareSerial mySerial(3, 2); /* RX:D3, TX:D2 */
ESP8266 wifi(mySerial);
void setup(void)
{
      Serial.begin(9600);
   mySerial.begin(9600);
    Serial.print("setup begin\r\n");
  Serial.print("FW Version: ");
    Serial.println(wifi.getVersion().c_str());
 if (wifi.setOprToStation()) {
        Serial.print("to station ok\r\n");
        
    } else {
        Serial.print("to station err\r\n");
          Serial.println("Wait 5 seconds and try again...");
            delay(5000);
    }
   while(1){
    if (wifi.joinAP(SSID, PASSWORD)) {
        Serial.print("Join AP success\r\n");
        Serial.print("IP: ");       
        Serial.println(wifi.getLocalIP().c_str());
     break;
    } else {
        Serial.print("Join AP failure\r\n");
        Serial.println("Wait 5 seconds and try again...");
            delay(5000);
    }
   
   }
           Serial.print("setup end\r\n");

}
void loop(void)
{
}

I have connected esp8266 to arduino uno as
    ESP8266_TX->RX1(D19)
    ESP8266_RX->TX1(D18)
    ESP8266_CH_PD->3.3V
    ESP8266_VCC->3.3V
    ESP8266_GND->GND.
setOprToStation() is returning False every time try to connect with the AP.

Please help me

rwkiii, 2016-03-04:
You may need to set both of your baud rates to 115200

agrawalnikhil, 2016-03-04:
I have both the baud rates to 115200 then also giving the same error

chinmoyrick, 2016-03-05:
what nothing is include. write about error.
one example .
#include <ESP8266WiFi.h>
#include <SPI.h>

Saums, 2016-04-05:
@agrawalnikhil, did you find the solution to this problem?

matheusaguiardev, 2017-02-17:
`mySerial.begin(115200);`
it's work for me !
into of setup

follow my example: 
`#include <SoftwareSerial.h>
 
//RX pino 2, TX pino 3
SoftwareSerial esp8266(2, 3);
 
#define DEBUG true
 
void setup()
{
  Serial.begin(9600);
  // Configure na linha abaixo a velocidade inicial do
  // modulo ESP8266
  esp8266.begin(115200);
  sendData("AT+RST\r\n", 2000, DEBUG);
  delay(1000);
  Serial.println("Versao de firmware");
  delay(3000);
  sendData("AT+GMR\r\n", 2000, DEBUG); // rst
  // Configure na linha abaixo a velocidade desejada para a
  // comunicacao do modulo ESP8266 (9600, 19200, 38400, etc)
  sendData("AT+CIOBAUD=9600\r\n", 2000, DEBUG);
  Serial.println("** Final **");
}
 
void loop() {}
 
String sendData(String command, const int timeout, boolean debug)
{
  // Envio dos comandos AT para o modulo
  String response = "";
  esp8266.print(command);
  long int time = millis();
  while ( (time + timeout) > millis())
  {
    while (esp8266.available())
    {
      // The esp has data so display its output to the serial window
      char c = esp8266.read(); // read the next character.
      response += c;
    }
  }
  if (debug)
  {
    Serial.print(response);
  }
  return response;
}`

matheusaguiardev, 2017-02-17:
Or.... 
```
// Carrega as bibliotecas ESP8266 e SoftwareSerial
#include "ESP8266.h"
#include "SoftwareSerial.h"
    
// Cria uma serial nas portas 2 (RX) e 3 (TX)
SoftwareSerial minhaSerial(2 , 3); 

// Define que o modulo ira utilizar a serial minhaSerial
ESP8266 wifi(minhaSerial);

// Configuracao ID e senha da rede Wireless
#define SSID        "NOME_DA_REDE"
#define PASSWORD    "SENHA_DA_REDE"

void setup()
{
    Serial.begin(9600);
    minhaSerial.begin(115200);
}

void loop(void)
{
    Serial.print("Inicializando modulo\r\n");
    Serial.print("Versao do firmware: ");
    Serial.println(wifi.getVersion().c_str());
    // Define modo de operacao como STA (station)
    if (wifi.setOprToStation()) {
        Serial.print("Modo STA ok\r\n");
    } else {
        Serial.print("Erro ao definir modo STA !r\n");
    }
    
    // Conexao a rede especificada em SSID
    if (wifi.joinAP(SSID, PASSWORD)) {
        Serial.print("Conectado com sucesso a rede wireless\r\n");
        Serial.print("IP: ");       
        Serial.println(wifi.getLocalIP().c_str());
    } else {
        Serial.print("Erro ao conectar rede wireless !!!\r\n");
    }
    
    Serial.print("*** Fim ***\r\n");
    while(1){}
}
```

myz3r0, 2017-03-10:
I have tired all the methods as suggested, however, I'm still getting that error.