itead/ITEADLIB_Arduino_WeeESP8266 #61: No response is printed

shadababe04, 2016-04-22:
nothing is printed in response like no FM version printed no ip printed in response..

setup begin

FW Version:

to station + softap ok

Join AP success

IP:

single ok

setup end

create tcp ok

release tcp ok

create tcp ok

DanieleDDM, 2016-09-04:
I have the same problem with a black 1MB esp8266. Which firmware is compatible with the library?

cesarcardinali, 2017-01-20:
Yeah, same here.
To be honest, it prints the response sometimes, but most of the times it doesn't. =(
I've noticed that it may be related to the received characters. Most of times it receives a bunch of messed up characters that are not part of UTF-8 or wherever.
Really sad.

beerygaz, 2017-09-17:
Are you using software serial?  Try slowing the baud rate right down to 19200 or 9600 (you need to use the AT+UART command to do this, try using the Serial Passthrough example in the Arduino library)

I'm still having issues myself, but if you're seeing a lot of gibberish characters it could be serial noise.