itead/ITEADLIB_Arduino_WeeESP8266 #69: Selected string

rafik73github, 2016-07-22:
Hello
Please help. I modified your code:
```
#include "ESP8266.h"
#include <ArduinoJson.h>

#define SSID        "net"
#define PASSWORD    "pass"
#define HOST_NAME   "192.168.1.8"
#define HOST_PORT   (8095)

ESP8266 wifi(Serial3);

void setup(void)
{
    Serial.begin(9600);
    Serial.print("setup begin\r\n");

    if (wifi.setOprToStation()) {
        Serial.print("to station ok\r\n");
    } else {
        Serial.print("to station err\r\n");
    }

    if (wifi.joinAP(SSID, PASSWORD)) {
        Serial.print("Join AP success\r\n");

        Serial.print("IP:");
        Serial.println( wifi.getLocalIP().c_str());       
    } else {
        Serial.print("Join AP failure\r\n");
    }
    
    if (wifi.disableMUX()) {
        Serial.print("single ok\r\n");
    } else {
        Serial.print("single err\r\n");
    }
    
    Serial.print("setup end\r\n");
}
 
void loop(void)
{
    uint8_t buffer[1024] = {0};
   

    if (wifi.createTCP(HOST_NAME, HOST_PORT)) {
        Serial.print("create tcp ok\r\n");
    } else {
        Serial.print("create tcp err\r\n");
    }

    char *hello = "GET /api HTTP/1.1\r\nHost: 192.168.1.8\r\n\r\n";
    wifi.send((const uint8_t*)hello, strlen(hello));

    uint32_t len = wifi.recv(buffer, sizeof(buffer), 10000);
    uint32_t i = 0;
    char json[len];
    if (len > 0 ) {
        //Serial.println("Pobrano:[");
       while(i < len){
         json[i++] = (char)buffer[i];    

        }
        
    }

      Serial.print(json);
   
/*      
StaticJsonBuffer<450> jsonBuffer;
  JsonObject& root = jsonBuffer.parseObject(json);
 
  if (!root.success()) {
   
Serial.print("blad JSON"); 
    return;
   }
String runningIndex = root["runningIndex"];
Serial.print("runningIndex: ");
Serial.print(runningIndex);
*/
   while(1);
    
}
```

monitor serial displays:

```
setup begin
to station ok
Join AP success
IP:+CIFSR:STAIP,"192.168.1.80"
+CIFSR:STAMAC,"a0:2a:81:c8:7c:00"
single ok
setup end
create tcp ok
TTP/1.1 200 OK
Content-Length: 32
Content-Type: application/json
Server: Microsoft-HTTPAPI/2.0
Access-Control-Allow-Origin: *
Date: Fri, 22 Jul 2016 07:19:21 GMT

{"lowVersion":1,"highVersion":4}
```
I want to display only:

`{"lowVersion":1,"highVersion":4}`

I need to read JSON.

chinmoyrick, 2016-07-22:
Hi..
"//" use this before    Serial.print("create tcp ok\r\n");
and
Serial.print("setup begin\r\n");
 Serial.print("create tcp ok\r\n");

rafik73github, 2016-07-23:
continues to be displayed:

```
TTP/1.1 200 OK
Content-Length: 32
Content-Type: application/json
Server: Microsoft-HTTPAPI/2.0
Access-Control-Allow-Origin: *
Date: Sat, 23 Jul 2016 05:16:56 GMT

{"lowVersion":1,"highVersion":4}
```

but I need only:

`{"lowVersion":1,"highVersion":4}`

You may need to modify the function **recvPkg** ?

chinmoyrick, 2016-07-23:
its reply your connection done and send data or recived.

chinmoyrick, 2016-07-23:
char json[len];

Serial.print(json);

Serial.print("blad JSON"); 

I think" Serial.print("blad JSON"); " use // before this and check....

LuizHAP, 2016-08-31:
chinmoyrick, i don't undestand the change. u can explain some more?

lenny1972, 2016-09-30:
you can use strimg.index of and Strig.substring functios to identify and select rhe only  part of the replay you need to display/use