itead/ITEADLIB_Arduino_WeeESP8266 #75: wifi.getVersion() and TCPServer example not working...

lenny1972, 2016-09-30:
I'm facing with two different issues on this library:
- I've tried to run the TCPServer example provided qith the library but after attaching to my Wi-Fi home network I didn't have any "signs of life" from the Arduino module via the Serial Monitor. 
Through the browser I cannot see any received message from the Serial Monitor when I try to call th two IP address provided with the wifi.getLocalIP() function.
I've tried also to ping both the ip address assigned by the router to the "AccessPoint" and to the "Station" (es. 192.168.1.xx and the 192.168.4.x) without any success.

In addition I've got not result when invoking the wifi.getVersion() function.

May someone out there help me?

Thanks in advanced

p.s.: I add no code because I reter to the library example sketches (the function wifi.getVersion() is on every example sketch)

stanely, 2016-11-09:
### My environment: Arduino UNO + ESP-01 boards.

- I found the communication between my arduino and ESP-01 board is working, somehow the text did not show as expected such as **getVersion()** . After do a research quickly I found the problem looks like is causing by "parsing function error". If you take a look at the implementation of this project can find easily, the library will parse the return string after issue a AT command. I believe the string sent from ESP-01 firmware is changed without notice so that this library is malfunction now.

- Since my purpose is to confirm this ESP-01 works, so I did not fix the problem here.

jcvsilva, 2016-11-21:
I had to change this part of the code in ESP8266.cpp, 
```
bool ESP8266::eATGMR(String &version)
{
    rx_empty();
    m_puart->println("AT+GMR");
    //return recvFindAndFilter("OK", "\r\r\n", "\r\n\r\nOK", version); 
    return recvFindAndFilter("OK", "+GMR", "\r\nOK", version);
}
```

marioaprea, 2020-03-01:
I changed all the recFindAndFilter response of each function