itead/ITEADLIB_Arduino_WeeESP8266 #79: "to station err, Join AP failure"

gsidsid, 2017-01-02:
Every time I run my modified ConnectWiFi example, I get the following:

setup begin
FW Version: 
to station err
Join AP failure

This seems to be quite a popular problem, and through all the previous answers I've adapted my code such that it should work. I am running my esp8266 with an Arduino Uno in the following config:

TX->D2
RX->D3
VCC->VCC
GPIO_0->VCC
GND->GND
RST->GND

Here is the code I am using. I modified it to use SoftwareSerial, set the baud rate to 115200, and obviously filled in the appropriate SSID and password. I recall uncommenting a line in the ESP8266 header file as well, for use with my uno. 

`#include "ESP8266.h"
#include <SoftwareSerial.h>

#define SSID        "SSID"
#define PASSWORD    "PASSWORD"

SoftwareSerial mySerial(3, 2); /* RX:D3, TX:D2 */
ESP8266 wifi(mySerial);

void setup(void)
{
    mySerial.begin(115200);
    Serial.begin(115200);
    Serial.print("setup begin\r\n");

    Serial.print("FW Version: ");
    Serial.println(wifi.getVersion().c_str());
    
    
    if (wifi.setOprToStation()) {
        Serial.print("to station ok\r\n");
    } else {
        Serial.print("to station err\r\n");
    }

    if (wifi.joinAP(SSID, PASSWORD)) {
        Serial.print("Join AP success\r\n");
        Serial.print("IP: ");       
        Serial.println(wifi.getLocalIP().c_str());
    } else {
        Serial.print("Join AP failure\r\n");
    }
    
    Serial.print(wifi.getLocalIP());
}

void loop(void)
{
}`

If anyone has any input to provide me as to what may be causing my problem, it would be greatly appreciated. Thanks.

jpatillo, 2017-03-02:
Try with these connections:

TX->D2
RX->D3
VCC->VCC
CH_PD->VCC
GND->GND

jdavies, 2017-03-12:
The example code omits the setting of the speed for Serial1. Simply adding 
`Serial1.begin(115200);`
near the top of the setup() method resolved the problem.