itead/ITEADLIB_Arduino_WeeESP8266 #86: Error: no matching function for call to 'ESP8266::ESP8266(SoftwareSerial*)'

Meldak, 2017-04-15:
Hi, i have this issue when a try to compile.

I read the documentation and I follow the instructions but when y try to create the serial connection with the serial libreri "SoftwareSerial.h" and read from my ESP8266 module, the compiler return me this error. 

I try to understand the file ESP8266.cpp but guess i'm doing this fine, some one can tell me what i doing wrong?

`#include "ESP8266.h"
#include <SoftwareSerial.h>

#define SSID        "ITEAD"
#define PASSWORD    "12345678"

SoftwareSerial esp(3,2); // RX | TX
ESP8266 wifi(esp);

void setup(void)
{
    Serial.begin(9600);
    Serial.print("setup begin\r\n");

    Serial.print("FW Version: ");
    Serial.println(wifi.getVersion().c_str());
    
    
    if (wifi.setOprToStation()) {
        Serial.print("to station ok\r\n");
    } else {
        Serial.print("to station err\r\n");
    }

    if (wifi.joinAP(SSID, PASSWORD)) {
        Serial.print("Join AP success\r\n");
        Serial.print("IP: ");       
        Serial.println(wifi.getLocalIP().c_str());
    } else {
        Serial.print("Join AP failure\r\n");
    }
    
    Serial.print("setup end\r\n");
}`

klren0312, 2017-05-03:
I alse have this problem.

christau, 2017-05-05:
Did you uncomment this [line](https://github.com/itead/ITEADLIB_Arduino_WeeESP8266/blob/master/ESP8266.h#L27) in ESP8266.h?
If not, this should be your problem.

SkywardCat, 2021-08-02:
I have the same problem. I have uncomment the line too.

Meldak, 2021-08-21:
Sorry for the late answer.

This problem is because I have two libraries installed and referenced in Arduino IDE, just uninstall one or delete the directory of one of them and update the package manager. 

Be careful when importing many libraries for a solution.

closed 2021-08-21 by Meldak