itead/ITEADLIB_Arduino_WeeESP8266 #89: Connection GET and build a project

ViictorG1, 2017-05-22:
Hi,

I'm needing help to build a project and I'm looking to use this library for my ESP8266 module.

I need the arduino reading the DB every time to receive a simple value 0 or 1, I just need to know how to receive this answer.

At the moment I am not able to make a code on the arduino with this library that makes the GET request work, could someone show me a code that would work? Thank you.

My PHP:
```
<?php
  $servername = "mysql.hostinger.com.br";
  $username = "root";
  $password = "*****";
  $dbname = "esp1";
  $conn = new mysqli($servername, $username, $password);
  
  header("content-type: application/json");

  $sql = "SELECT  * FROM `u582151594_esp82`.`esp1`";
  $result = $conn->query($sql);
  while($data = $result->fetch_object()) {
   $value = $data->value;      
  }
  
  echo '{"value":';
  echo $value;
  echo '}';
  
  $conn->close();
?>
```
The code returns me:
{"value":0}

I need the arduino to read this JSON, parse, and remove this VALUE for me to use in my code.

Some help?

Thankss

igorleessa, 2017-05-23:
Take a look at the library, 
https://github.com/interactive-matter/aJson/tree/master/Examples

igorleessa, 2017-05-23:
https://bblanchon.github.io/ArduinoJson/doc/decoding/