itead/ITEADLIB_Arduino_WeeESP8266 #93: "udp send" methods does not work my esp8266

masaokubo, 2017-06-23:
I use Arduino uno r3 with esp8266 firm 0018000902.
I have tried UDP packet from this uno to its sever on another pc by "send" method in ITEADLIB_Arduino_WeeESP8266.
However it output no packets.
Could you give me any suggestions.

thank you for your support.

masaokubo, 2017-06-24:
My code is described as follows.
JoinAP, registerUDP is ok but this does not send any packets.

thank you for your supports.

-----------------------------------------------
#include "ESP8266.h"

#define SSID        "XXXXX"
#define PASSWORD    "XXXXX"
#define HOST_NAME   "XXXX"
#define HOST_PORT  (YYYY)

SoftwareSerial mySerial(11, 12); /* RX:D12, TX:D11 */
ESP8266 wifi(mySerial);
void setup(void)
{
    Serial.begin(9600);  
    Serial.print("setup begin\r\n");
    
    Serial.print("FW Version:");
    Serial.println(wifi.getVersion().c_str());
      
    if (wifi.setOprToStationSoftAP()) {
        Serial.print("to station + softap ok\r\n");
    } else {
        Serial.print("to station + softap err\r\n");
    }
 
    if (wifi.joinAP(SSID, PASSWORD)) {
        Serial.print("Join AP success\r\n");
        Serial.print("IP: ");
        Serial.println(wifi.getLocalIP().c_str()); 
             
    } else {
        Serial.print("Join AP failure\r\n");
    }
    
    if (wifi.disableMUX()) {
        Serial.print("single ok\r\n");
    } else {
        Serial.print("single err\r\n");
    }
    
    Serial.print("setup end\r\n");
    
}
void loop(void)
{
    uint8_t buffer[128] = {0};
    
   if (wifi.registerUDP(HOST_NAME, HOST_PORT)) {
        Serial.print("register udp ok\r\n");
    } else {
        Serial.print("register udp err\r\n");
    }
    
    char msgxx[256];
    sprintf(msgxx,"Hello, this is client!\r\n");
    wifi.send((const uint8_t*)msgxx, strlen(msgxx));
    Serial.println(msgxx);

    if (wifi.unregisterUDP()) {
        Serial.print("unregister udp ok\r\n");
    } else {
        Serial.print("unregister udp err\r\n");
    }
  
    
    delay(5000);
}

vicjomaa, 2018-02-27:
Hello I`ve the same problem using Arduino Uno R3, indeed the console show me this messages when i probe your code.
unregister udp err
register udp err

 could you fix the problem ?
thanks for all...