itead/ITEADLIB_Arduino_WeeESP8266 #94: ESP8266 not responding with atmega 2560 or Arduino uno

creatvipin, 2017-08-31:
[image]
I have used ESP8266 SDK version 2.0.1 latest with STM32F401,STM32F407 .But it does not respond to AT Command with STM32 or any other microcontroller.

I have tested it with ITEAD with arduino mega and Arduino uno available library link is given below.
https://github.com/Pranavgulati/ESP8266/blob/master/README.md
In this i have used SDK 0.18 and 0.22 version and latest version 2.0.1 but it does not respond to any of the command from MCU whereas it is working fine with usb to serial converter using terminal.
I have used the bidirectional level shifter also between Arduino and ESP8266 .
Supply current rating is also 1A.

So i want to know that whether how could i can make it work using the Other MCU apart from terminal.

I have searched lot of place,but i did not find any satisfactory answer and applied all the approaches to make it work.I have used ESP8266 SDK version 2.0.1 latest with STM32F401,STM32F407 .But it does not respond to AT Command with STM32 or any other microcontroller.

I have tested it with ITEAD with arduino mega and Arduino uno available library link is given below.
https://github.com/Pranavgulati/ESP8266/blob/master/README.md
In this i have used SDK 0.18 and 0.22 version and latest version 2.0.1 but it does not respond to any of the command from MCU whereas it is working fine with usb to serial converter using terminal.
I have used the bidirectional level shifter also between Arduino and ESP8266 .
Supply current rating is also 1A.

So i want to know that whether how could i can make it work using the Other MCU apart from terminal.

I have searched lot of place,but i did not find any satisfactory answer and applied all the approaches to make it work.

I am also attaching the circuit i used and the error i am getting
[image]

Jamiek94, 2017-12-20:
You could try the following:

- If you are using software serial, set the baud rate to 9600 (within the esp chip & in code)
- Replace this method in the library:

```
bool ESP8266::eATGMR(String &version)
{
    rx_empty();
    m_puart->println("AT+GMR");
    return recvFindAndFilter("OK", "+GMR", "\r\nOK", version); // new
    //return recvFindAndFilter("OK", "\r\r\n", "\r\n\r\nOK", version);  // old
}
```

habi3000, 2018-04-20:
hi, i think your problem is with the hardware connections, the mega and the board need to have a common ground. it is not clear in the circuit you provided if they have or not.