itead/ITEADLIB_Arduino_WeeESP8266 #96: wifi.recv returns zero length

Jamiek94, 2017-12-20:
I'm having some issues receiving data back from wifi. I can connect to an access point just fine, creating a tcp connection also works. But after that it starts to fail.

I'm using an Arduino Uno with a scheme like this, except I am using Software Serial: 

https://cdn.instructables.com/FC5/FW5J/IGHOIQAH/FC5FW5JIGHOIQAH.LARGE.jpg

My code looks like the following:

`#include "ESP8266.h"

#define SSID        "Fapcave"
#define PASSWORD    "tnkt6455"
#define HOST_NAME   "www.baidu.com"
#define HOST_PORT   (80)

SoftwareSerial esp8266(2, 3);
ESP8266 wifi(esp8266);

void setup(void)
{
    Serial.print("Setup\r\n");
    esp8266.begin(9600);
    Serial.begin(9600);

    Serial.print("FW Version:");
    Serial.println(wifi.getVersion().c_str());
    
    if (wifi.joinAP(SSID, PASSWORD)) {
        delay(2000);
        Serial.print("Join AP success\r\n");

        Serial.print("IP:");
        Serial.println( wifi.getLocalIP().c_str());    
    } else {
        Serial.print("Join AP failure\r\n");
    }

    delay(2000);
    
    if (wifi.disableMUX()) {
        Serial.print("single ok\r\n");
    } else {
        Serial.print("single err\r\n");
    }
    
    Serial.print("setup end\r\n");

    delay(5000);
    
    uint8_t buffer[126] = {0};

    if (wifi.createTCP(HOST_NAME, HOST_PORT)) {
        Serial.print("create tcp ok\r\n");
    } else {
        Serial.print("create tcp err\r\n");
    }

    delay(5000);

    Serial.print("sending request\r\n");
    
    char *hello = "GET / HTTP/1.1\r\nHost: www.baidu.com\r\nConnection: close\r\n\r\n";
    wifi.send((const uint8_t*)hello, strlen(hello));

    delay(5000);

    Serial.print("receiving request\r\n");

    uint32_t len = wifi.recv(buffer, sizeof(buffer), 10000);

    Serial.print("request length:");
    Serial.println(len);
    
    if (len > 0) {
        Serial.print("Received:[");
        for(uint32_t i = 0; i < len; i++) {
            Serial.print((char)buffer[i]);
        }
        Serial.print("]\r\n");
    }

    if (wifi.releaseTCP()) {
        Serial.print("release tcp ok\r\n");
    } else {
        Serial.print("release tcp err\r\n");
    }
}
 
void loop(void)
{
    delay(100000);
}
`

Serial monitor output;

FW Version:

`
AT version:1.2.0.0(Jul  1 2016 20:04:45)
SDK version:1.5.4.1(39cb9a32)
Ai-Thinker Technology Co. Ltd.
Dec  2 2016 14:21:16
Join AP success
IP:+CIFSR:STAIP,"192.168.43.192"
+CIFSR:STAMAC,"5c:cf:7f:b2:96:29"
single ok
setup end
create tcp ok
sending request
receiving request
request length:0
release tcp err
`

I did change a method in the library, I had to do this otherwise it wouldn't print anything to the serial monitor.

`
bool ESP8266::eATGMR(String &version)
{
    rx_empty();
    m_puart->println("AT+GMR");
    return recvFindAndFilter("OK", "+GMR", "\r\nOK", version); // new
    //return recvFindAndFilter("OK", "\r\r\n", "\r\n\r\nOK", version);  // old
}
`

Any ideas what's causing this issue? I've tried several websites.

yoller, 2018-10-13:
Did u resolve this? I have the same problem!

Jamiek94, 2018-10-19:
I'm pretty sure I got it working by requesting a response from a website which only displays a simple text file with only a line of text.

It had something to do with the size of the buffer. Take a look at the attention part here: https://github.com/itead/ITEADLIB_Arduino_WeeESP8266/blob/master/README.md

lilfish, 2019-03-25:
Also having this issue. I'm using an express server localhost to post data to. The data get's posted, but ESP8266 doesn't seem to get a response. Even with a simple "OK" or "1" it doesnt receive anything. Ideas?

Stslit, 2022-06-12:
Help! Same question!