itead/ITEADLIB_Arduino_WeeESP8266 #100: Not Building With Wemos D1 Mini

RayyanTahir, 2018-04-29:
Hi, I'm using Wemos D1 mini development board [https://wiki.wemos.cc/products:d1:d1_mini_lite](https://wiki.wemos.cc/products:d1:d1_mini_lite).

Whenever I build this library with that board selected, I get the following errors:

expected identifier before numeric constant
ESP8266.h:38: note  in expansion of macro ESP8266
   class ESP8266 {

expected unqualified-id before numeric constant
ESP8266.h:38: note  in expansion of macro ESP8266
   class ESP8266 {
 
   expected unqualified-id before numeric constant
ESP8266.cpp:51: note  in expansion of macro ESP8266
 ESP8266*: ESP8266(HardwareSerial &uart, uint32_t baud): m_puart(&uart)
 
   expected unqualified-id before numeric constant
ESP8266.cpp:58: note  in expansion of macro ESP8266
 bool ESP8266*: kick(void)
 
   expected unqualified-id before numeric constant
ESP8266.cpp:63: note  in expansion of macro ESP8266
 bool ESP8266*: restart(void)
Error compiling project sources

The library builds fine if I have an Arduino board selected (specifically Arduino Uno).

per1234, 2018-04-29:
The reason is that the class name conflicts with the ESP8266 macro defined by the ESP8266 core.

I have a strong suspicion you don't understand the purpose of this library. It's supposed to be used with an Arduino board which is using an ESP8266 module running the AT firmware as a WiFi adapter. The only time it would make sense to compile this library for the WeMos D1 Mini or any other ESP8266 board is if you had two ESP8266 boards connected together and it would be a very rare case when that would make any sense.

RayyanTahir, 2018-05-01:
Thank you @per1234 for your reply. I was actually trying to add the src to my library project (in Visual Studio IDE for Arduino) where I could add/modify code in my own git repo. Anyhow I went on without compiling it myself.

closed 2018-05-01 by RayyanTahir